themactep/thingino-firmware #740: Record to SD card eventually fails to work

Nord221, 2025-08-24:
Hi, I've encountered this issue three times now. It takes about half a month to a month to occur.

What happens is that old recordings get deleted (as expected) but the folder containing the recordings is still there. Eventually the recordings will only save on the present day and then stop completely.

I've SSH into my device and retrieved the following df and dh results. From the results it appears something is not getting deleted properly as there is a discrepancy between df and du. The df command shows 100% usage of the 32 GB SD card but du shows only 1.9 GB. 

Please let me know if there are anything additional that would be of assistance. I've already swapped with a brand new SD card (Samsung Endurance) but the issue occurred again so it might not an SD card issue. The old SD card I've tested on my computer and it is able to be reformatted and read/written to properly.

FYI current version is master+596d148, 2025-07-22 09:12:23 +0000 but this issue has happened months prior on the latest update for that month.

```
root@ing-wyze-cam3-0730 ing-wyze-cam3-0730# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 6.1M      6.1M         0 100% /rom
devtmpfs                 46.5M         0     46.5M   0% /dev
/dev/mtdblock2          224.0K    120.0K    104.0K  54% /overlay
overlay                 224.0K    120.0K    104.0K  54% /
/dev/mtdblock5            8.0M    260.0K      7.8M   3% /opt
tmpfs                    46.6M     32.0K     46.5M   0% /dev/shm
tmpfs                    46.6M    228.0K     46.3M   0% /tmp
tmpfs                    46.6M     44.0K     46.5M   0% /run
/dev/mmcblk0p1           29.8G     29.8G     28.0M 100% /mnt/mmcblk0p1
```

```
root@ing-wyze-cam3-0730 ing-wyze-cam3-0730# du -h records/
1.9G	records/20250823
64.0K	records/20250807
64.0K	records/20250808
64.0K	records/20250809
32.0K	records/20250810
1.9G	records/
```

sstepansky, 2025-08-24:
Maybe the processes that created the files are still running and have the file descriptors locked open.
Use `lsof` and `ps` to check that possibility.

sstepansky, 2025-08-24:
Mmm looks like it's not what I thought...
You could also try `sync`...

Nord221, 2025-08-24:
I tried `sync` along with rebooting but unfortunately it did not have any effect on the SD card storage.

virmaior, 2025-08-26:
is the card formatted exfat or fat32?

virmaior, 2025-08-26:
I had entire batch of exfat 128gb samsung endurance go bad relatively quickly. 
I switched to generic 32gb fat32 cards and haven't seen the same issues.

I think I had the sort of issue you're describing _once_. More commonly I saw cards having read/write issues and/or corruption especially when full.

Nord221, 2025-08-26:
Was the entire batch of exFAT 128GB Samsung Endurance cards that went bad used only in Thingino devices or was it other devices?

So far from my experience these cards are very reliable as I have them in my dashcam and Amcrest cameras and it has been 5+ years without problems.

Only in Thingino do I encounter read/write or corruption issues. And like you mentioned, they typically happen when the device becomes full. Reformatting the card resolves the issue.

Nord221, 2025-08-27:
Appreciate your input on this. Just wanted to add Wyze's official RTSP firmware is also reliable in terms RTSP and SD card albeit very outdated. Hopefully this can be resolved in a future update.

Nord221, 2025-09-08:
Other than the checks I posted above there are no logs to be provided unfortunately. Let me know if there are logs you'd like to review.

lifeishard, 2025-09-10:
I am pretty sure I encountered the same issue on new version and previous versions of thingino. 32gb SD card, formatted in exfat using thingino webui interface. 

version master+f35dbd6, 2025-09-08 17:50:17 +0000

> /dev/mmcblk0p1           29.8G     29.8G     15.8M 100% /mnt/mmcblk0p1
> 
> SD card mounts
> # grep ^/dev/mmc /proc/mounts
> 
> /dev/mmcblk0p1 /mnt/mmcblk0p1 exfat rw,sync,noatime,fmask=0000,dmask=0000,allow_utime=0022,namecase=0,errors=remount-ro 0 0
> 

I deliberately set the storage limit to be 28gb, but this problem still occurs. The recording to SD card failed all of a sudden and just stopped. Subsequent cam reboots just leads to constantly writing 0 byte files to the sd card, which is a common error I've noticed.

root@ing-wyze-cam2-8B67 ing-wyze-cam2-8B67# du -h
32.0K   ./records/20250824
32.0K   ./records/20250825
32.0K   ./records/20250826
32.0K   ./records/20250827
32.0K   ./records/20250828
32.0K   ./records/20250829
32.0K   ./records/20250830
32.0K   ./records/20250831
32.0K   ./records/20250901
32.0K   ./records/20250317
32.0K   ./records/20250902
32.0K   ./records/20250903
32.0K   ./records/20250904
32.0K   ./records/20250905
32.0K   ./records/20250906/21
32.0K   ./records/20250906/22
32.0K   ./records/20250906/23
128.0K  ./records/20250906
32.0K   ./records/20250907
180.8M  ./records/20250908/17
29.7M   ./records/20250908/18
497.7M  ./records/20250908/21
680.5M  ./records/20250908/22
520.6M  ./records/20250908/23
1.9G    ./records/20250908
859.2M  ./records/20250909/00
829.2M  ./records/20250909/01
699.5M  ./records/20250909/02
618.1M  ./records/20250909/03
602.4M  ./records/20250909/04
633.4M  ./records/20250909/05
913.7M  ./records/20250909/06
924.9M  ./records/20250909/07
850.4M  ./records/20250909/08
930.3M  ./records/20250909/09
857.9M  ./records/20250909/10
940.3M  ./records/20250909/11
839.1M  ./records/20250909/12
692.9M  ./records/20250909/13
513.5M  ./records/20250909/14
530.1M  ./records/20250909/15
564.9M  ./records/20250909/16
560.1M  ./records/20250909/17
681.3M  ./records/20250909/18
887.7M  ./records/20250909/19
921.5M  ./records/20250909/20
899.1M  ./records/20250909/21
824.4M  ./records/20250909/22
824.5M  ./records/20250909/23
18.0G   ./records/20250909
727.8M  ./records/20250910/00
240.7M  ./records/20250910/05
933.7M  ./records/20250910/06
739.8M  ./records/20250910/07
563.4M  ./records/20250910/08
679.7M  ./records/20250910/09
709.8M  ./records/20250910/10
722.1M  ./records/20250910/11
739.3M  ./records/20250910/12
457.8M  ./records/20250910/13
409.7M  ./records/20250910/16
525.8M  ./records/20250910/17
67.4M   ./records/20250910/18
7.3G    ./records/20250910
27.2G   ./records
27.2G   .

root@ing-wyze-cam2-8B67 ing-wyze-cam2-8B67# df -T
Filesystem           Type       1K-blocks      Used Available Use% Mounted on
/dev/root            squashfs        6016      6016         0 100% /rom
devtmpfs             devtmpfs       45456         0     45456   0% /dev
/dev/mtdblock2       jffs2            224       148        76  66% /overlay
overlay              overlayfs        224       148        76  66% /
/dev/mtdblock5       jffs2           8320       260      8060   3% /opt
tmpfs                tmpfs          45552        24     45528   0% /dev/shm
tmpfs                tmpfs          45552      2436     43116   5% /tmp
tmpfs                tmpfs          45552       116     45436   0% /run
/dev/mmcblk0p1       exfat       31272096  31255968     16128 100% /mnt/mmcblk0p1

virmaior, 2025-09-10:
out of curiosity, what are you settings? 
You're using 9gb/day; I'm usingabout 4gb/day  in h264 and 3.2gb in h265 on  on wyze3 and atom2 recording in 1 minute segments.

lifeishard, 2025-09-10:
My diagnostic logs http://tb.thingino.com/dx5z/

Wyze2 cam, everything is default, except audio is AAC, since for some reason opus is kind of bugged, I recall there was a issue raised about the audio before.

lifeishard, 2025-09-18:
I reformatted my card and encountered the issue again on master+0ae27f0, 2025-09-10 21:09:25 +0000. I reviewed the system logs which were on debug mode at the time the recordings stopped and didn't see anything related. After the recording stopped the two record processes were still running.

> 18582     1 root     S     1976  2.1   0  0.0 {record} /bin/sh /sbin/record
> 19681 18582 root     S     1976  2.1   0  0.0 {record} /bin/sh /sbin/record

The recording process stopped writing to sd card around 4am, interestingly it somehow wrote one more recording file at 9 am.

Nord221, 2025-09-20:
I format my SD card outside of Thingino and whether it is actually exFAT or FAT32 Thingino will always display it as exFAT.

This is a behavior I notice over the past month of testing:

If the SD card is formatted as exFAT then the recordings will stop after a period of 1 to 2 weeks.

However, if the SD card is formatted as FAT32 then the SD card will become corrupted after a period of 1 to 2 weeks. When the SD card is corrupted all files on the card will become unreadable and looks like the following screenshot:

[image]

Nord221, 2025-09-20:
@virmaior 
1. 5 minutes
2. 32 GB (Samsung Pro Endurance)

Also I have the Storage Limit on the recording settings set to 28 GB, though this doesn't seem to make any difference.

virmaior, 2025-09-20:
interesting.

I had the cards dying when I was doing:
1. 1 minute
2. 128 GB Samsung Pro Endurance

Since the cards were dying I swtiched to cheapo generic 32 GB cards (I think Kyocera under the hood) from a legitimate seller and basically haven't encountered the same issues.

virmaior, 2025-09-21:
@lifeishard  I should work things more carefully.

1. If I check the cards in a different device, they don't show up as having errors (with I think one exception).
2. When using 128GB cards, I was getting the filename corruption problem described by @Nord221 is his comment yesterday.
3. What I see more recently is a log with jzmmc events. I don't remember the text of the message but there's a message that says there's a hardware problem with the SD card or reader and then the card won't write any more.
4. For cards that are starting to fritz, messages like will start to appear:
```
[7387190.271040] jzmmc_v1.2 jzmmc_v1.2.0: timeout 1000ms op:25 w sz:62976 state:2 STAT:0x1E000580 DMALEN:0x00007E00 blks:59/123 clk:enable clk_gate:enable
[7389303.033504] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:63488 state:2 STAT:0x1E000580 DMALEN:0x0000C000 blks:27/124 clk:enable clk_gate:enable
[7389303.533529] jzmmc_v1.2 jzmmc_v1.2.0: timeout 1000ms op:25 w sz:63488 state:2 STAT:0x1E000580 DMALEN:0x00007E00 blks:60/124 clk:enable clk_gate:enable
[7389304.033530] jzmmc_v1.2 jzmmc_v1.2.0: timeout 1500ms op:25 w sz:63488 state:2 STAT:0x1E000580 DMALEN:0x00003C00 blks:93/124 clk:enable clk_gate:enable
[7390020.921938] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:64512 state:2 STAT:0x1E000580 DMALEN:0x00008600 blks:58/126 clk:enable clk_gate:enable
[7390021.421903] jzmmc_v1.2 jzmmc_v1.2.0: timeout 1000ms op:25 w sz:64512 state:2 STAT:0x1E000580 DMALEN:0x00004600 blks:90/126 clk:enable clk_gate:enable
[7390021.921920] jzmmc_v1.2 jzmmc_v1.2.0: timeout 1500ms op:25 w sz:64512 state:2 STAT:0x1E000580 DMALEN:0x00000400 blks:123/126 clk:enable clk_gate:enable
```

(this type of message appears when it's having trouble finding enough room to write)