themactep/thingino-firmware #749: Enabling day to night contol by sun tracking and disabling it breaks attempts to use "by illumination" later

lcwsoft, 2025-09-05:
I noticed something odd recently. I have a Wyze v3 camera in a completely dark room. Testing, I first tried to set the day to night control to use sun tracking, which unsurprisingly didn't work well for me as it would be too dark in the room during the day to see anything. I then tried setting day to night control by illumination, which itself works well. However I noticed that every day, it would stop working in the same way (couldn't see anything). 

Steps to reproduce:

1) Install thingino on a camera and put it in a completely dark room. While I can only test a Wyze v3, it is possible that this affects other cameras as well.

ASSERT: the video feed from the camera should be dark with a hint of purple.

2) If you enable sun tracking for day night control under Settings > Day/Night Mode Control with any valid lat and long coordinates, it will create a set of cron jobs like the following:

```
# see https://github.com/themactep/thingino-firmware/wiki/Cron-jobs

# reboot camera nightly @ 3:00
#0 3 * * * reboot -f
# run daynight every 1 minutes
#*/1 * * * * daynight
# run dusk2dawn nightly at 0:00
0 0 * * * dusk2dawn
# switch to day mode at sunrise (autogenerated)
24 06 * * * daynight day
# switch to night mode at sunset (autogenerated)
34 19 * * * daynight night
```

So far so good.

3) If you disable sun tracking under Settings > Day/Night Mode Control, the cron jobs are changed to the following:

```
# see https://github.com/themactep/thingino-firmware/wiki/Cron-jobs

# reboot camera nightly @ 3:00
#0 3 * * * reboot -f
# run dusk2dawn nightly at 0:00
0 0 * * * dusk2dawn
# switch to day mode at sunrise (autogenerated)
# switch to night mode at sunset (autogenerated)
# run daynight every 1 minutes
#*/1 * * * * daynight
```

4) If you then enable day night changes by illumination under Settings > Day/Night Mode Control, the cron jobs are changed to the following:

```
# see https://github.com/themactep/thingino-firmware/wiki/Cron-jobs

# reboot camera nightly @ 3:00
#0 3 * * * reboot -f
# run dusk2dawn nightly at 0:00
0 0 * * * dusk2dawn
# switch to day mode at sunrise (autogenerated)
# switch to night mode at sunset (autogenerated)
# run daynight every 1 minutes
*/1 * * * * daynight
```

ASSERT: in a completely dark room with day mode and night mode gains of 500 and 15000 respectively (the defaults), the camera should switch to night mode in 1 minute.

5) At midnight, dusk2dawn runs and disables the daynight cron job that is used for checking using illumination:

```
# see https://github.com/themactep/thingino-firmware/wiki/Cron-jobs

# reboot camera nightly @ 3:00
#0 3 * * * reboot -f
# run dusk2dawn nightly at 0:00
0 0 * * * dusk2dawn
# switch to day mode at sunrise (autogenerated)
24 06 * * * daynight day
# switch to night mode at sunset (autogenerated)
34 19 * * * daynight night
```

ASSERT: if you check after the time for "daynight day", you should see that the video is now dark again.

6) Go back to Settings > Day/Night Mode Control and click the save button

ASSERT: dusk2dawn is still enabled

```
# see https://github.com/themactep/thingino-firmware/wiki/Cron-jobs

# reboot camera nightly @ 3:00
#0 3 * * * reboot -f
# run dusk2dawn nightly at 0:00
0 0 * * * dusk2dawn
# switch to day mode at sunrise (autogenerated)
# switch to night mode at sunset (autogenerated)
# run daynight every 1 minutes
*/1 * * * * daynight
```

This means that it will run again at the next midnight and the same issue will occur again in the morning.

This breaks thingino usage with cameras in a completely dark room (as you would need to use night mode for the video feed to be usable) if you mistakenly enable sun tracking at any time, and then switch to the illumination method.

Proposed Resolution:

dusk2dawn appears to be solely for setting the cron jobs for day and night mode changes when using the sun tracking method. Therefore the dusk2dawn command should check the status of the sun tracking setting and do nothing if it is disabled. In addition, disabling sun tracking should disable the dusk2dawn cron job.

Workaround:

Click the "cron" link on the Settings > Day/Night Mode Control page and comment out the cron job for the dusk2dawn command.

virmaior, 2025-09-19:
I'm really not a fan of the web interface reading the interval value for daynight from the crontab file.

The dawn2dusk and daynight crons can basically end up competing and the web interface presents them somewhat confusingly.　

basically I put in a pull request that would address this among other things but there's better things on the horizon that will make my pull request obsolete (and there's some question as to whether using a continuous do loop as I did is a bad idea).

themactep, 2026-01-16:
Irrelevant with the new photosensing in the streamer.

closed 2026-01-16 by themactep