themactep/thingino-firmware #755: Sonoff cam slim gen2 noisy audio

limawaken, 2025-09-08:
audio is only static-like noise.
tried reducing input gain but audio becomes too soft and hardly audible.

no issues with factory firmware rtsp stream.

using firmware master+cbb7b68, 2025-08-29 08:56:35

limawaken, 2025-09-08:
setting noise suppression level to 3 reduces a lot of the noise, but audio is very soft.

themactep, 2025-09-21:
starting from today, all updates will be built from stable branch. it is less fancy but battletested. please update and confirm it's working for you.

m4mmon, 2025-09-22:
@limawaken a guitar pick would be a good friend
You can "easily" insert the pick between the black and white part. Working towards a corner, at some point the corner would pop like in the picture. From there, you can pop the other corners. Be gentle, there are some wires.

[image]

Initially the red and black wires were running on the left of the lens. Like I said I accidentally discovered that if it was in the position like in the picture, the noise would not be there anymore.

[image]

Hope it helps.