themactep/thingino-firmware #757: Wyze Cam v3 not working with USB Ethernet adapter

dbro, 2025-09-09:
Hello, thanks for Thingino, it is working very well for me so far. Now I am trying to get two Wyze Cam v3 devices to use a wired connection instead of the built in wireless. I am using an OTG adapter and have tested two different USB network adapters, on both of the two devices, and see the same error message in dmesg. Here are some lines from dmesg that might be relevant.

```
[    0.728334] usbcore: registered new interface driver asix
[    0.734215] usbcore: registered new interface driver cdc_ether
[    0.740553] usbcore: registered new interface driver cdc_ncm

...

[    0.869156] usb 1-1: device descriptor read/64, error -88
[    1.169173] usb 1-1: device descriptor read/64, error -88
[    1.211900] jffs2: notice: (407) jffs2_build_xattr_subsystem: complete building xattr subsystem, 2 of xdatum (0 unchecked, 2 orphan) and 2 of xref (0 dead, 2 orphan) found.
[    1.286465] jffs2: notice: (419) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[    1.469170] usb 1-1: new high-speed USB device number 3 using dwc2
[    1.669173] usb 1-1: device descriptor read/64, error -88
[    1.969130] usb 1-1: device descriptor read/64, error -88
[    2.269135] usb 1-1: new high-speed USB device number 4 using dwc2
[    2.689101] usb 1-1: device not accepting address 4, error -88
[    2.879125] usb 1-1: new high-speed USB device number 5 using dwc2
[    3.299109] usb 1-1: device not accepting address 5, error -88
[    3.305159] hub 1-0:1.0: unable to enumerate USB device on port 1
```

Both adapters are 10+ years old and work fine on a laptop running Debian linux. One of the network adapters is made by Apple, and uses the Asix AX88772A chip, and the Debian system successfully loads the modules called ax88796b and asix to use it. The other adapter is a generic adapter, and the Debian system recognizes it as MOSCHIP 7830/7832/7730 and then loads module mcs7830 for it. I noticed in the [Thingino wiki page about USB networking](https://github.com/themactep/thingino-firmware/wiki/USB-Ethernet-Networking) that thingino supports "ASIX AX88772-based USB adapters" , which seems to match the Apple adapter I'm trying to use.

How can I make progress troubleshooting this? I didn't see anything in the logread or logcat that mentioned "USB" "net" or "asix".

Please let me know what I can try next, thanks!

themactep, 2025-09-17:
[image] 

[image]

[image]

themactep, 2025-09-17:
i don't have a 88772A adapter but 88772E works fine.

themactep, 2025-09-17:
[image]

88772C works too.

themactep, 2025-09-17:
[image]

and 88772A too (found one)

dbro, 2025-09-17:
Thank you very much for this suggestion. I tried some different cables and it works now!

Sorry to have not tried this earlier, the documentation suggests that some cables might work better than others.