themactep/thingino-firmware #771: reading prudynt stream is somehow broken with 2025-09-18

m4mmon, 2025-09-18:
Hello,

Some software have problems reading streams from latest release (2025-09-18 as of writing).
Complete default prudynt settings after a "sysupgrade -f" on a Eufy T8410c.
Only things set are root password and timezone.

vlc : sometimes the video will play for 2, 3 or 4 seconds and stop. Sometimes when it starts reading the stream, I can hear audio coming from camera, and it stops before the video freezes. Switching to TCP in VLC settings does not make any difference.
Sometimes only a single frame is shown, from the beginning.
Sometimes the video will not freeze, sound is out-of-sync, and in VLC UI "time" indicator is jumping back and forth to values like "10:39:14" and "46:41". Uptime being a few minutes.

mpv: sometimes video can be fine, but audio is out of sync and comes 1/2 seconds. If I restart prudynt or mpv, audio might come too early, or too late. Sometimes audio will stop like:
```
AO: [pulse] 48000Hz stereo 2ch float
VO: [wlshm] 1920x1080 yuv420p
AV: 00:00:03 / 09:51:27 (0%) A-V:  0.000
Invalid audio PTS: 4.060000 -> 35484.781167
Reset playback due to audio timestamp reset.
AV: 00:00:12 / 09:51:35 (0%) A-V:  0.000
```
Timestamp looks weird, uptime was around a few minutes when I got this log.

Here, after restarting mpv a few seconds later, audio comes early:
```
AO: [pulse] 48000Hz stereo 2ch float
VO: [wlshm] 1920x1080 yuv420p
Invalid video timestamp: 0.240778 -> 0.040889
AV: 00:00:02 / 03:25:32 (0%) A-V:  0.002 ct:  0.198 Dropped: 1
Invalid video timestamp: 2.880833 -> 12330.618311

Audio/Video desynchronisation detected! Possible reasons include too slow
hardware, temporary CPU spikes, broken drivers, and broken files. Audio
position will not match to the video (see A-V status field).

AV: 03:25:39 / 03:25:43 (99%) A-V:-12324.072 ct: -3.466 Dropped: 1
```

ffplay/ffmpeg seems to be the most reliable and it looks like everything is fine. But it is quite long to start and switches by itself to TCP after UDP timeout, something I don't remember having seen before.
```
[rtsp @ 0x76ef34000bc0] UDP timeout, retrying with TCP   0B f=0/0
Input #0, rtsp, from 'rtsp://thingino:thingino@192.168.161.101:554/ch0':
  Metadata:
    title           : thingino prudynt
    comment         : stream0
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(tv, bt709, progressive), 1920x1080, 25 fps, 25 tbr, 90k tbn, 50 tbc
    Stream #0:1: Audio: opus, 48000 Hz, stereo, fltp
  42.67 A-V:-12382.244 fd=   0 aq=    9KB vq=   11KB sq=    0B f=1/1
```

I don't see anything funny in the logs.

Sorry if it is a lot of info.
I can do tests etc. if asked.

dekonnection, 2025-09-18:
Same thing here with 2025-09-18 and 2025-09-17 (at least) on a Jooan A2R : the stream generally only shows a still image, sometimes a few seconds of stream before stopping.

dekonnection, 2025-09-18:
Can confirm that 2025-09-13 works, if that helps.

Breakage may have been introduced with 2025-09-17.

dekonnection, 2025-09-19:
Still broken for me on Jooan A2R with `master+a5c812e` / `2025-09-19`.

m4mmon, 2025-09-20:
Hi,
Here are some logs and prudynt config file. I booted the camera, no web interface, ssh-ed into it, tee-ed logread, logcat and tried three times to read the stream with VLC.

eufy_t8410c_t31x_sc3338_syn4343
master+a5c812e, 2025-09-19 09:11:04 +0000

[logs_and_config.zip](https://github.com/user-attachments/files/22438910/logs_and_config.zip)

Charky-Dev, 2025-09-20:
I also experienced this issue with the latest release (for Aoqee C1) - the videos were all corrupted. 
Settings would have been default.
Installing the version from the 13th worked fine.

m4mmon, 2025-09-20:
@Charky-Dev @dekonnection yes I did not mention it, but the firmwares from the 13th are fine for me too.

dekonnection, 2025-09-20:
Still broken with today's 2025-09-20.

themactep, 2025-09-20:
Downgrade to older build and wait till things are sorted out.

m4mmon, 2025-09-20:
If I can offer any help, tests, etc. just tell me what I can do.

Charky-Dev, 2025-09-20:
Unfortunately my older firmware has now also started corrupting the video footage - it worked for about an hour. Also happy to provide logs etc if you can explain how to do so; the usual methods I would try don't work.

Anaerin, 2025-09-20:
Had the same problem in a Wyze v2 camera (the bottom half of the video stream was cut off and smeared), and changing
```
"rtsp": {
[...]
  "out_buffer_size": 300000,
  "send_buffer_size": 253600,
[...]
},
```
from their original values of `200000` and `153600` solved my video smearing and packet drop issues.

Anaerin, 2025-09-21:
From what I was seeing, the output packets were getting truncated (that is, cut short), so depending on the bitrate of the video I would get more or less of the video image, and doing things like disabling audio would get slightly more. It might be worth looking at previous versions and seeing if that value has changed, or if something else related is happening.

themactep, 2025-09-21:
starting from today, all updates will be built from stable branch. it is less fancy but battletested. please update and confirm it's working for you.

m4mmon, 2025-09-22:
Sorry for the delay, just tested 2025-09-22, works fine :)