themactep/thingino-firmware #774: Request:  Set to defaults on Streamer/OSD page

davlaw, 2025-09-21:
After messing with these setting for a few weeks , would like to get it back to default with out a re-flash.

While it might just be time to bite the bullet and re-flash.  Not sure how much memory consumption would be involved, it might be prohibitive just to satisfy a creature comfort.

m4mmon, 2025-09-21:
take that as "to be confirmed by more experienced people":
SSH-ing to the camera, I would simply:
```
# destroy the modified version of the file
rm /overlay/etc/prudynt.cfg
# force a remount
mount -o remount /
# restart prudynt
service restart prudynt
```

It is also possible to restore complete system defaults without reflashing from the web interface, using settings/reset/reset firmware.
or apparently, from ssh, doing a "firstboot -f"

themactep, 2025-09-28:
Information -> Commands and logs -> prudynt, the bottom at the bottom of the page below the config itself. It should be there in tomorrow builds. 

[image]