themactep/thingino-firmware #782: stable 2025-09-27 - webui messes prudynt.json

m4mmon, 2025-09-27:
Hi,

wuuk_y0510_t31x_sc4336p_ssv6158
stable+3fc63ae, 2025-09-27 01:52:21 +0000

Install from scratch.
Using the "Save config" button in "Settings" / "Streamer/OSD" in the webui leads to repetitions in prudynt.json.
Not all keys seem to be duplicated, I guess it depends on what is sent from the browser when the click occurs.

For example, after modifying streamer settings and saving a few times:

```
        },
        "stream2":      {
                "enabled":      true,
                "jpeg_channel": 0,
                "jpeg_idle_fps":        1,
                "jpeg_path":    "/tmp/snapshot.jpg",
                "jpeg_quality": 75,
                "jpeg_refresh": 1000,
                "enabled":      true,
                "jpeg_path":    "/tmp/snapshot.jpg",
                "jpeg_channel": 0,
                "jpeg_quality": 75,
                "jpeg_idle_fps":        1,
                "fps":  25,
                "enabled":      true,
                "jpeg_path":    "/tmp/snapshot.jpg",
                "jpeg_channel": 0,
                "jpeg_quality": 75,
                "jpeg_idle_fps":        1,
                "fps":  25,
                "enabled":      true,
                "jpeg_path":    "/tmp/snapshot.jpg",
                "jpeg_channel": 0,
                "jpeg_quality": 75,
                "jpeg_idle_fps":        1,
                "fps":  25
        },
```

Thanks!

m4mmon, 2025-09-28:
I confirm it does not happen with 2025-09-28 :)

themactep, 2025-09-29:
many isp settings from the form are meant to work with imp-control, not directly with prudynt. that is an outdated approach, leftover from openipc majestic streamer times which is closed, so we could not implement that in the streamer directly. we need to integrate handling of these settings in prudynt directly, but only if we had time for everything. checkout sources, dig in, improve, send patches.