themactep/thingino-firmware #786: sysupgrade check_free_space() fails with misleading error message with suggestion to run gestalt if image file not found during nightly build

inactivist, 2025-09-29:
If running `sysupgrade -p` during nightly build, and the asset has not yet been built resulting in a 404 error when trying to retrieve the image size, check_free_space fails with a misleading error message and suggestion:

```
Cannot get remote file size. Camera profile name may have changed.
Please run gestalt and select the correct profile.
```

The suggestion to run `gestalt` is unhelpful as the system profile name is in fact correct and running `gestalt` to choose the correct profile does not resolve the problem.

This is due to failure to determine image file size since the asset isn't available.

If you wait a while for the asset to be available in the asset list web page, the `sysupgrade -p` command will succeed.

Example:

```sh
~# sysupgrade -p
Run as /sbin/sysupgrade -p
Downloading firmware into /tmp, there is enough free space.
Create /tmp/sysupgrade directory
Update self
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade to /tmp/sysupgrade/sysupgrade
###################################################################################################################################################################################### 100.0%
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade-stage2 to /sbin/sysupgrade-stage2
###################################################################################################################################################################################### 100.0%
Re-run script from /tmp/sysupgrade
Run as /tmp/sysupgrade/sysupgrade -p
Downloading firmware into /tmp, there is enough free space.
Check connection to https://github.com
Change directory to /tmp
Upgrade from GitHub
Cannot get remote file size. Camera profile name may have changed.
Please run gestalt and select the correct profile.
Clean up temporary files.
```

During an in-progress nightly build (GitHub release page showed `Release: firmware-2025-09-29 is currently being built, please wait...` the target image was not shown in the asset list.  

I used the same `curl` command issued in `check_free_space()` (with the image name for my Cinnado D1 t23 as computed by the `sysupgrade` script):

```sh
~# curl -sIL https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-cinnado_d1_t23n_sc2336_atbm6012bx.bin
HTTP/1.1 302 Found
Date: Mon, 29 Sep 2025 09:20:36 GMT
Content-Type: text/html; charset=utf-8
Content-Length: 0
Vary: X-PJAX, X-PJAX-Container, Turbo-Visit, Turbo-Frame, X-Requested-With,Accept-Encoding, Accept, X-Requested-With
Location: https://github.com/themactep/thingino-firmware/releases/download/firmware-2025-09-29/thingino-cinnado_d1_t23n_sc2336_atbm6012bx.bin
Cache-Control: no-cache
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-Frame-Options: deny
X-Content-Type-Options: nosniff
X-XSS-Protection: 0
Referrer-Policy: no-referrer-when-downgrade
Content-Security-Policy: default-src 'none'; base-uri 'self'; child-src github.githubassets.com github.com/assets-cdn/worker/ github.com/assets/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com www.githubstatus.com collector.github.com raw.githubusercontent.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com *.rel.tunnels.api.visualstudio.com wss://*.rel.tunnels.api.visualstudio.com github.githubassets.com objects-origin.githubusercontent.com copilot-proxy.githubusercontent.com proxy.individual.githubcopilot.com proxy.business.githubcopilot.com proxy.enterprise.githubcopilot.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com productionresultssa0.blob.core.windows.net/ productionresultssa1.blob.core.windows.net/ productionresultssa2.blob.core.windows.net/ productionresultssa3.blob.core.windows.net/ productionresultssa4.blob.core.windows.net/ productionresultssa5.blob.core.windows.net/ productionresultssa6.blob.core.windows.net/ productionresultssa7.blob.core.windows.net/ productionresultssa8.blob.core.windows.net/ productionresultssa9.blob.core.windows.net/ productionresultssa10.blob.core.windows.net/ productionresultssa11.blob.core.windows.net/ productionresultssa12.blob.core.windows.net/ productionresultssa13.blob.core.windows.net/ productionresultssa14.blob.core.windows.net/ productionresultssa15.blob.core.windows.net/ productionresultssa16.blob.core.windows.net/ productionresultssa17.blob.core.windows.net/ productionresultssa18.blob.core.windows.net/ productionresultssa19.blob.core.windows.net/ github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com wss://alive.github.com wss://alive-staging.github.com api.githubcopilot.com api.individual.githubcopilot.com api.business.githubcopilot.com api.enterprise.githubcopilot.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com copilot-workspace.githubnext.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: blob: github.githubassets.com media.githubusercontent.com camo.githubusercontent.com identicons.github.com avatars.githubusercontent.com private-avatars.githubusercontent.com github-cloud.s3.amazonaws.com objects.githubusercontent.com release-assets.githubusercontent.com secured-user-images.githubusercontent.com/ user-images.githubusercontent.com/ private-user-images.githubusercontent.com opengraph.githubassets.com marketplace-screenshots.githubusercontent.com/ copilotprodattachments.blob.core.windows.net/github-production-copilot-attachments/ github-production-user-asset-6210df.s3.amazonaws.com customer-stories-feed.github.com spotlights-feed.github.com objects-origin.githubusercontent.com *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/ secured-user-images.githubusercontent.com/ private-user-images.githubusercontent.com github-production-user-asset-6210df.s3.amazonaws.com gist.github.com; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; upgrade-insecure-requests; worker-src github.githubassets.com github.com/assets-cdn/worker/ github.com/assets/ gist.github.com/assets-cdn/worker/
Server: github.com
09:21:04 GMT; Secure; SameSite=Lax
Set-Cookie: logged_in=no; Path=/; Domain=github.com; Expires=Tue, 29 Sep 2026 09:21:04 GMT; HttpOnly; Secure; SameSite=Lax

HTTP/1.1 404 Not Found
Date: Mon, 29 Sep 2025 09:20:36 GMT
Content-Type: text/plain; charset=utf-8
Content-Length: 9
Vary: X-PJAX, X-PJAX-Container, Turbo-Visit, Turbo-Frame, X-Requested-With,Accept-Encoding, Accept, X-Requested-With
Cache-Control: no-cache
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-Frame-Options: deny
X-Content-Type-Options: nosniff
X-XSS-Protection: 0
Referrer-Policy: no-referrer-when-downgrade
Content-Security-Policy: default-src 'none'; base-uri 'self'; connect-src 'self'; form-action 'self'; img-src 'self' data:; script-src 'self'; style-src 'unsafe-inline'
Server: github.com
```

_(some irrelevant headers removed from output)_

gtxaspec, 2025-10-02:
sysupgrade should notify to the actual curl error (http error code?), the current notice:

```
Cannot get remote file size. Camera profile name may have changed.
Please run gestalt and select the correct profile.
```
is a bit misleading, notes for the future.

themactep, 2025-10-05:
fixed with updated workflow

closed 2025-10-05 by themactep